mu-editor/mu #2424: Setting up development environment takes forever

paulvanharen, 2023-06-28:
### What were you trying to do?

Setup my computer for Mu development

### What steps did you take to trigger the issue?

1. Clone the git repo
2. Create a virtual environment
3. Activate the environment
4. Run `pip install -e ".[dev]"` as per documentation

### What did you expect to happen?

I expect pip to finish in a reasonable time, under 1 hr

### What actually happened?

Please find the output by pip below. Pip is still running and I trust there is some marginal progress...
Please note the INFO messages reading

    INFO: This is taking longer than usual. You might need to provide the dependency resolver with stricter constraints to reduce runtime. See https://pip.pypa.io/warnings/backtracking for guidance. If you want to abort this run, press Ctrl + C.

There are quite a number of these.

Obtaining file:///home/paul/workspace/mu
  Preparing metadata (setup.py) ... done
Collecting adafruit-board-toolkit~=1.1
  Downloading adafruit_board_toolkit-1.1.1-py3-none-any.whl (12 kB)
Collecting click<=8.0.4
  Downloading click-8.0.4-py3-none-any.whl (97 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 97.5/97.5 kB 4.6 MB/s eta 0:00:00
Collecting flake8>=3.8.3
  Using cached flake8-6.0.0-py2.py3-none-any.whl (57 kB)
Collecting ipykernel<6,>=4.1
  Downloading ipykernel-5.5.6-py3-none-any.whl (121 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 121.0/121.0 kB 11.5 MB/s eta 0:00:00
Collecting ipython_genutils>=0.2.0
  Downloading ipython_genutils-0.2.0-py2.py3-none-any.whl (26 kB)
Collecting jupyter-client<6.2,>=4.1
  Downloading jupyter_client-6.1.12-py3-none-any.whl (112 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 112.7/112.7 kB 12.6 MB/s eta 0:00:00
Collecting nudatus>=0.0.3
  Downloading nudatus-0.0.5-py3-none-any.whl (5.3 kB)
Collecting platformdirs<3.0.0,>=2.0.0
  Using cached platformdirs-2.6.2-py3-none-any.whl (14 kB)
Collecting pyserial~=3.5
  Downloading pyserial-3.5-py2.py3-none-any.whl (90 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 90.6/90.6 kB 9.6 MB/s eta 0:00:00
Collecting qtconsole==4.7.7
  Downloading qtconsole-4.7.7-py2.py3-none-any.whl (118 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 118.5/118.5 kB 11.8 MB/s eta 0:00:00
Collecting requests>=2.0.0
  Using cached requests-2.31.0-py3-none-any.whl (62 kB)
Collecting semver>=2.8.0
  Downloading semver-3.0.1-py3-none-any.whl (17 kB)
Collecting virtualenv>=20.0.0
  Downloading virtualenv-20.23.1-py3-none-any.whl (3.3 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 3.3/3.3 MB 10.8 MB/s eta 0:00:00
Collecting wheel
  Using cached wheel-0.40.0-py3-none-any.whl (64 kB)
Collecting PyQt5==5.13.2
  Downloading PyQt5-5.13.2-5.13.2-cp35.cp36.cp37.cp38-abi3-manylinux1_x86_64.whl (62.2 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 62.2/62.2 MB 7.3 MB/s eta 0:00:00
Collecting PyQtChart==5.13.1
  Downloading PyQtChart-5.13.1-5.13.2-cp35.cp36.cp37.cp38-abi3-manylinux1_x86_64.whl (1.1 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.1/1.1 MB 10.6 MB/s eta 0:00:00
Collecting QScintilla==2.11.3
  Downloading QScintilla-2.11.3-cp35.cp36.cp37.cp38-abi3-manylinux1_x86_64.whl (2.7 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 2.7/2.7 MB 2.6 MB/s eta 0:00:00
Collecting black<22.1.0,>=19.10b0
[... 676 lines cut ...]
INFO: pip is looking at multiple versions of pyserial to determine which version is compatible with other requirements. This could take a while.
INFO: This is taking longer than usual. You might need to provide the dependency resolver with stricter constraints to reduce runtime. See https://pip.pypa.io/warnings/backtracking for guidance. If you want to abort this run, press Ctrl + C.
Using cached platformdirs-2.6.0-py3-none-any.whl (14 kB)
  Downloading platformdirs-2.5.4-py3-none-any.whl (14 kB)
  Downloading platformdirs-2.5.3-py3-none-any.whl (14 kB)
  Downloading platformdirs-2.5.2-py3-none-any.whl (14 kB)
  Downloading platformdirs-2.5.1-py3-none-any.whl (14 kB)
  Downloading platformdirs-2.5.0-py3-none-any.whl (14 kB)
INFO: pip is looking at multiple versions of platformdirs to determine which version is compatible with other requirements. This could take a while.
  Downloading platformdirs-2.4.1-py3-none-any.whl (14 kB)
  Downloading platformdirs-2.4.0-py3-none-any.whl (14 kB)
  Downloading platformdirs-2.3.0-py3-none-any.whl (13 kB)
  Downloading platformdirs-2.2.0-py3-none-any.whl (13 kB)

### Operating System Version

Ubuntu - 23.04

### Mu Version

Latest from git

### Other Info

Python version is 3.11.2

paulvanharen, 2023-06-29:
I tried again using Python version 3.9.16. Unfortunately, with similar results: `pip._vendor.resolvelib.resolvers.ResolutionTooDeep: 2000000`

carlosperate, 2023-06-30:
Unfortunatelly Mu is not yet compatible with Python 3.9 or newer, we are still stuck on some old dependencies we are planning to upgrade as soon as we get Mu v1.2.1 out.

If you can try with Python 3.8 do you get the same issue?

paulvanharen, 2023-07-04:
@carlosperate: Thanks for highlighting. This indeed has been the issue. I've installed another Python version from Deadsnakes and now it works fine.

closed 2023-07-04 by paulvanharen