mu-editor/mu #2454: Release 1.2.0 linux

Juniorsnet, 2023-08-08:
The appimage on github release page is not the same on TAR file.

sha256 of appimage in TAR file from relese page = 10726cec6d1fa947337b53b6cdc19e0f3b689b51997cc721c9c903e984483f3c

sha256 of appimage from release page = 1a302a772dcc60b6353e44e4cc772a37fec7238c573bbb88d01ed7f6fcbf6abe

The file from release page crash with segfault (corrupted?)

The appimage from TAR file works OK

carlosperate, 2023-08-10:
Hi @Juniorsnet,

Thanks for the report. If I remember correctly the .tar file fixed something compared to the tar one (maybe it was just the executable flag), so that should be the main file to download.

@ntoll we should probably delete the `*.AppImage` file from the release page as we should only ask users to download the  .tar one.

haijinSk, 2023-08-17:
I can confirm the issue on Ubuntu 22.04.3. In my case, the appimage on github release page simply does not run (I allowed executing file as program); the appimage from TAR file works.