mu-editor/mu #2456: Ctrl-V paste doesn't work in Ubuntu install

wyzarddoc, 2023-08-25:
### What were you trying to do?

paste a .PY copy into Mu

### What steps did you take to trigger the issue?

Highlighted the test from the program then right clicked selected paste and nothing

### What did you expect to happen?

expected the copied text to be placed in the editor

### What actually happened?

nothing 

### Operating System Version

Ubuntu 22.04

### Mu Version

1.2

### Editor Log

2023-08-25 13:20:51,149 - root:283(run) INFO: 

-----------------

Starting Mu 1.2.0
2023-08-25 13:20:51,150 - root:284(run) INFO: uname_result(system='Linux', node='doc-dt', release='6.2.0-26-generic', version='#26~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Jul 13 16:27:29 UTC 2', machine='x86_64', processor='x86_64')
2023-08-25 13:20:51,154 - root:285(run) INFO: Platform: Linux-6.2.0-26-generic-x86_64-with-glibc2.2.5
2023-08-25 13:20:51,154 - root:286(run) INFO: Python path: ['/tmp/.mount_Mu_EdiyhFo8Y/usr/lib/python38.zip', '/tmp/.mount_Mu_EdiyhFo8Y/usr/lib/python3.8', '/tmp/.mount_Mu_EdiyhFo8Y/usr/lib/python3.8/lib-dynload', '/tmp/.mount_Mu_EdiyhFo8Y/usr/lib/python3.8/site-packages']
2023-08-25 13:20:51,154 - root:287(run) INFO: Language code: en_US
2023-08-25 13:20:51,154 - mu.settings:220(load) WARNING: No settings file found at /home/doc/.local/share/mu/settings.json; skipping
2023-08-25 13:20:51,155 - mu.settings:220(load) WARNING: No settings file found at /home/doc/.local/share/mu/session.json; skipping
2023-08-25 13:20:51,155 - root:312(run) INFO: Wayland detected, setting QT_QPA_PLATFORM=wayland
2023-08-25 13:20:51,504 - mu.virtual_environment:619(ensure_and_create) INFO: Added log handler.
2023-08-25 13:20:51,505 - mu.virtual_environment:630(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2023-08-25 13:20:51,505 - mu.virtual_environment:837(create_venv) INFO: Creating virtualenv: /home/doc/.local/share/mu/mu_venv-38-20230825-132048
2023-08-25 13:20:51,505 - mu.virtual_environment:838(create_venv) INFO: Virtualenv name: mu_venv-38-20230825-132048
2023-08-25 13:20:51,506 - mu.virtual_environment:463(run_subprocess) INFO: Running ('/tmp/.mount_Mu_EdiyhFo8Y/usr/bin/python3.8', '-I', '-m', 'virtualenv', '-p', '/tmp/.mount_Mu_EdiyhFo8Y/usr/bin/python3.8', '-q', '--symlinks', '/home/doc/.local/share/mu/mu_venv-38-20230825-132048') with kwargs {'env': {'USER': 'doc', 'XDG_SESSION_TYPE': 'wayland', 'SHLVL': '0', 'DESKTOPINTEGRATION': 'AppImageLauncher', 'HOME': '/home/doc', 'APPDIR': '/tmp/.mount_Mu_EdiyhFo8Y', 'DESKTOP_SESSION': 'ubuntu', 'REDIRECT_APPIMAGE': '/home/doc/Applications/Mu_Editor-1.2.0-x86_64_74e65f527f143e0143de8e9e10ed41a4.AppImage', 'GTK_MODULES': 'gail:atk-bridge', 'GNOME_SHELL_SESSION_MODE': 'ubuntu', 'MANAGERPID': '2410', 'SYSTEMD_EXEC_PID': '2583', 'DBUS_SESSION_BUS_ADDRESS': 'unix:path=/run/user/1000/bus,guid=7e638432ac6fc152e993b73864e8dc5d', 'DBUS_STARTER_BUS_TYPE': 'session', 'GIO_LAUNCHED_DESKTOP_FILE_PID': '12771', 'IM_CONFIG_PHASE': '1', 'WAYLAND_DISPLAY': 'wayland-0', 'LOGNAME': 'doc', 'TCL_LIBRARY': '/tmp/.mount_Mu_EdiyhFo8Y/usr/lib/tcl8.6', 'JOURNAL_STREAM': '8:33958', '_': '/usr/bin/gnome-session', 'OWD': '/home/doc/Downloads/python/pyusb', 'XDG_SESSION_CLASS': 'user', 'USERNAME': 'doc', 'GNOME_DESKTOP_SESSION_ID': 'this-is-deprecated', 'PATH': '/home/doc/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/pico/pico-sdk:/snap/bin', 'SESSION_MANAGER': 'local/doc-dt:@/tmp/.ICE-unix/2583,unix/doc-dt:/tmp/.ICE-unix/2583', 'INVOCATION_ID': '7a31f195fb8347cf9e6dbb6f930d8810', 'APPIMAGE': '/home/doc/Applications/Mu_Editor-1.2.0-x86_64_74e65f527f143e0143de8e9e10ed41a4.AppImage', 'GNOME_SETUP_DISPLAY': ':1', 'XDG_RUNTIME_DIR': '/run/user/1000', 'XDG_MENU_PREFIX': 'gnome-', 'DISPLAY': ':0', 'XDG_CURRENT_DESKTOP': 'ubuntu:GNOME', 'LANG': 'en_US.UTF-8', 'XMODIFIERS': '@im=ibus', 'XAUTHORITY': '/run/user/1000/.mutter-Xwaylandauth.VEPEA2', 'XDG_SESSION_DESKTOP': 'ubuntu', 'SSH_AGENT_LAUNCHER': 'gnome-keyring', 'SSH_AUTH_SOCK': '/run/user/1000/keyring/ssh', 'SHELL': '/bin/bash', 'ARGV0': '/home/doc/Applications/Mu_Editor-1.2.0-x86_64_74e65f527f143e0143de8e9e10ed41a4.AppImage', 'QT_ACCESSIBILITY': '1', 'GDMSESSION': 'ubuntu', 'QT_IM_MODULE': 'ibus', 'PWD': '/home/doc/Downloads/python/pyusb', 'DBUS_STARTER_ADDRESS': 'unix:path=/run/user/1000/bus,guid=7e638432ac6fc152e993b73864e8dc5d', 'XDG_DATA_DIRS': '/usr/share/ubuntu:/home/doc/.local/share/flatpak/exports/share:/var/lib/flatpak/exports/share:/usr/local/share/:/usr/share/:/var/lib/snapd/desktop', 'XDG_CONFIG_DIRS': '/etc/xdg/xdg-ubuntu:/etc/xdg', 'QT_API': 'pyqt5', 'QT_AUTO_SCREEN_SCALE_FACTOR': '1', 'QT_MAC_WANTS_LAYER': '1', 'QT_QPA_PLATFORM': 'wayland'}}
2023-08-25 13:20:54,378 - mu.virtual_environment:475(run_subprocess) DEBUG: Process returned 0; output: No output received.
2023-08-25 13:20:54,379 - mu.virtual_environment:857(create_venv) INFO: Created virtual environment using /tmp/.mount_Mu_EdiyhFo8Y/usr/bin/python3.8 at /home/doc/.local/share/mu/mu_venv-38-20230825-132048
2023-08-25 13:20:54,379 - mu.virtual_environment:934(install_baseline_packages) INFO: Installing baseline packages.
2023-08-25 13:20:54,379 - mu.virtual_environment:139(run_blocking) INFO: About to run blocking /home/doc/.local/share/mu/mu_venv-38-20230825-132048/bin/pip with args ['--version', '--disable-pip-version-check'] and envvars {}
2023-08-25 13:20:57,867 - mu.virtual_environment:189(wait) DEBUG: Finished: True; exitStatus 0; exitCode 0
2023-08-25 13:20:57,867 - mu.virtual_environment:274(run) DEBUG: Process output: pip 22.3.1 from /home/doc/.local/share/mu/mu_venv-38-20230825-132048/lib/python3.8/site-packages/pip (python 3.8)
2023-08-25 13:20:57,867 - mu.virtual_environment:935(install_baseline_packages) INFO: pip version: pip 22.3.1 from /home/doc/.local/share/mu/mu_venv-38-20230825-132048/lib/python3.8/site-packages/pip (python 3.8)
2023-08-25 13:20:57,867 - mu.virtual_environment:942(install_baseline_packages) INFO: Expecting zipped wheels at /tmp/.mount_Mu_EdiyhFo8Y/usr/lib/python3.8/site-packages/mu/wheels/1.2.0.zip
2023-08-25 13:20:58,723 - mu.virtual_environment:916(install_from_zipped_wheels) INFO: About to install from wheel: matplotlib_inline-0.1.6-py3-none-any.whl
2023-08-25 13:20:58,732 - mu.virtual_environment:139(run_blocking) INFO: About to run blocking /home/doc/.local/share/mu/mu_venv-38-20230825-132048/bin/pip with args ['install', '--disable-pip-version-check', '--no-deps', '--no-index', '/tmp/tmpu1tre4si/matplotlib_inline-0.1.6-py3-none-any.whl'] and envvars {}
2023-08-25 13:21:01,555 - mu.virtual_environment:189(wait) DEBUG: Finished: True; exitStatus 0; exitCode 0
2023-08-25 13:21:01,556 - mu.virtual_environment:274(run) DEBUG: Process output: Processing /tmp/tmpu1tre4si/matplotlib_inline-0.1.6-py3-none-any.whl
Installing collected packages: matplotlib-inline
Successfully installed matplotlib-inline-0.1.6
2023-08-25 13:21:01,556 - mu.virtual_environment:916(install_from_zipped_wheels) INFO: About to install from wheel: click-8.1.3-py3-none-any.whl
2023-08-25 13:21:01,556 - mu.virtual_environment:139(run_blocking) INFO: About to run blocking /home/doc/.local/share/mu/mu_venv-38-20230825-132048/bin/pip with args ['install', '--disable-pip-version-check', '--no-deps', '--no-index', '/tmp/tmpu1tre4si/click-8.1.3-py3-none-any.whl'] and envvars {}
2023-08-25 13:21:04,577 - mu.virtual_environment:189(wait) DEBUG: Finished: True; exitStatus 0; exitCode 0
2023-08-25 13:21:04,577 - mu.virtual_environment:274(run) DEBUG: Process output: Processing /tmp/tmpu1tre4si/click-8.1.3-py3-none-any.whl
Installing collected packages: click
Successfully installed click-8.1.3
2023-08-25 13:21:04,577 - mu.virtual_environment:916(install_from_zipped_wheels) INFO: About to install from wheel: Flask-2.0.3-py3-none-any.whl
2023-08-25 13:21:04,577 - mu.virtual_environment:139(run_blocking) INFO: About to run blocking /home/doc/.local/share/mu/mu_venv-38-20230825-132048/bin/pip with args ['install', '--disable-pip-version-check', '--no-deps', '--no-index', '/tmp/tmpu1tre4si/Flask-2.0.3-py3-none-any.whl'] and envvars {}
2023-08-25 13:21:06,321 - mu.virtual_environment:189(wait) DEBUG: Finished: True; exitStatus 0; exitCode 0
2023-08-25 13:21:06,321 - mu.virtual_environment:274(run) DEBUG: Process output: Processing /tmp/tmpu1tre4si/Flask-2.0.3-py3-none-any.whl
Installing collected packages: Flask
Successfully installed Flask-2.0.3
[... cut ...]
pickleshare       0.7.5
pip               22.3.1
platformdirs      2.5.4
prompt-toolkit    3.0.32
ptyprocess        0.7.0
pure-eval         0.2.2
pygame            2.1.2
Pygments          2.13.0
python-dateutil   2.8.2
pyzmq             24.0.1
setuptools        65.5.1
six               1.16.0
stack-data        0.6.1
tornado           6.2
traitlets         5.5.0
wcwidth           0.2.5
Werkzeug          2.2.2
wheel             0.38.4
2023-08-25 13:36:25,756 - mu.virtual_environment:993(installed_packages) INFO: []

ProgramminCat, 2023-08-25:
Which program did you copy the text from? And is it raw text in your clipboard?

wyzarddoc, 2023-08-26:
Tried copying text from Geany and Thonny  both text editors /Gui's did not check for clipboard but have cut and pasted from these programs into many other editorsunder Linux
Web-design, Woodworking, 3D-Art