mu-editor/mu #2469: installing GObject module fails

Axel-Erfurt, 2023-11-13:
### What were you trying to do?

installing GObject

### What steps did you take to trigger the issue?

added GObject to Third Party Packages 

### What did you expect to happen?

installing GObject

### What actually happened?

Error: error: 
[Errno 2] No such file or directory: 'clang'

### Operating System Version

LMDE6 (Debian 12)

### Mu Version

1.2.0 Appimage

### Other Info

```
Collecting pygobject
Using cached PyGObject-3.46.0.tar.gz (723 kB)
Installing build dependencies: started
Installing build dependencies: finished with status 'error'
error: subprocess-exited-with-error
  
  × pip subprocess to install build dependencies did not run successfully.
  │ exit code: 1
  ╰─> [41 lines of output]
      Collecting setuptools
        Using cached setuptools-68.2.2-py3-none-any.whl.metadata (6.3 kB)
      Collecting wheel
        Using cached wheel-0.41.3-py3-none-any.whl.metadata (2.2 kB)
      Collecting pycairo
        Using cached pycairo-1.25.1.tar.gz (347 kB)
        Installing build dependencies: started
        Installing build dependencies: finished with status 'done'
        Getting requirements to build wheel: started
        Getting requirements to build wheel: finished with status 'done'
        Installing backend dependencies: started
        Installing backend dependencies: finished with status 'done'
        Preparing metadata (pyproject.toml): started
        Preparing metadata (pyproject.toml): finished with status 'done'
      Using cached setuptools-68.2.2-py3-none-any.whl (807 kB)
      Using cached wheel-0.41.3-py3-none-any.whl (65 kB)
      Building wheels for collected packages: pycairo
        Building wheel for pycairo (pyproject.toml): started
        Building wheel for pycairo (pyproject.toml): finished with status 'error'
        error: subprocess-exited-with-error
      
        × Building wheel for pycairo (pyproject.toml) did not run successfully.
        │ exit code: 1
        ╰─> [11 lines of output]
            running bdist_wheel
            running build
            running build_py
            creating build
            creating build/lib.linux-x86_64-cpython-38
            creating build/lib.linux-x86_64-cpython-38/cairo
            copying cairo/__init__.py -> build/lib.linux-x86_64-cpython-38/cairo
            copying cairo/__init__.pyi -> build/lib.linux-x86_64-cpython-38/cairo
            copying cairo/py.typed -> build/lib.linux-x86_64-cpython-38/cairo
[... 639 lines cut ...]
python-dateutil   2.8.2
pyzmq             24.0.1
setuptools        65.6.3
six               1.16.0
stack-data        0.6.1
tornado           6.2
traitlets         5.5.0
wcwidth           0.2.5
Werkzeug          2.2.2
wheel             0.38.4
2023-11-13 22:31:52,665 - mu.virtual_environment:993(installed_packages) INFO: ['PyQt5', 'PyQt5-Qt5', 'PyQt5-sip', 'PyQtWebEngine', 'PyQtWebEngine-Qt5']
2023-11-13 22:31:58,495 - mu.logic:1513(sync_package_state) INFO: Synchronize package states...
2023-11-13 22:31:58,495 - mu.logic:1514(sync_package_state) INFO: Old: {'pyqtwebengine-qt5', 'pyqt5', 'pyqt5-qt5', 'pyqtwebengine', 'pyqt5-sip'}
2023-11-13 22:31:58,495 - mu.logic:1515(sync_package_state) INFO: New: {'pyqtwebengine-qt5', 'pyqt5', 'pygobject', 'pyqt5-qt5', 'pyqtwebengine', 'pyqt5-sip'}
2023-11-13 22:31:58,495 - mu.logic:1519(sync_package_state) INFO: To add: {'pygobject'}
2023-11-13 22:31:58,495 - mu.logic:1520(sync_package_state) INFO: To remove: set()
2023-11-13 22:31:58,495 - mu.logic:1521(sync_package_state) INFO: Virtualenv: /home/brian/.local/share/mu/mu_venv
2023-11-13 22:31:58,499 - mu.virtual_environment:967(install_user_packages) INFO: Installing user packages: pygobject
2023-11-13 22:31:58,500 - mu.virtual_environment:152(run) INFO: About to run /home/brian/.local/share/mu/mu_venv-38-20231112-194901/bin/pip with args ['install', '--disable-pip-version-check', '--upgrade', 'pygobject'] and envvars {}
2023-11-13 22:32:14,168 - mu.settings:169(save) DEBUG: Saving to /home/brian/.local/share/mu/session.json
2023-11-13 22:32:14,169 - mu.logic:1430(quit) INFO: Quitting.

2023-11-13 22:32:14,169 - mu.settings:169(save) DEBUG: Saving to /home/brian/.local/share/mu/session.json
2023-11-13 22:32:14,170 - mu.settings:169(save) DEBUG: Saving to /home/brian/.local/share/mu/venv.json
```

carlosperate, 2023-12-11:
Hi @Axel-Erfurt 

Looks like PyGObject is only distributed as a source package, which means it needs to be built every time it is pip installed.

Do you have the build tools installed and available in the PATH? If you run `pip install PyGObject` in a virtual environment outside of Mu, does it build and install correctly?

Axel-Erfurt, 2023-12-12:
> If you run `pip install PyGObject` in a virtual environment outside of Mu, does it build and install correctly?

Yes