mu-editor/mu #2471: Version 1.2 for Mac OS as .dmg

mac-arno, 2023-11-22:
This Version does not start on Mac OS „High Sierra“ 10.13.6.

carlosperate, 2023-12-11:
Thanks for the report @mac-arno. It's been a while, but I thought I had tested this with macOS 10.13. Could you share a screenshot of the error?

mac-arno, 2023-12-14:
There is nothing to see, when I double click on the app icon of v1.2 after copying it into the Application folder. There is apparently nothing visible to happen. Not even in the system log.

carlosperate, 2023-12-27:
Thanks for the response @mac-arno!
That's unfortunate, normally the crash reporter should kick in and wrap the error message to help us debug it.

Could you check if you can find the log in this directory and attach it here?
`~/Library/Logs/mu/mu.log`

mac-arno, 2024-01-09:
There is no log output in the advised directory from starting version 1.2. Even in the Activity application there is no running process pointing to Mu. It seems that Mu tries to start and quits very silently.

carlosperate, 2024-01-11:
Can you try the macOs test build from this page?
https://github.com/mu-editor/mu/actions/runs/7342861479
[image]

As those installers aren't signed you'll have to right click on the .app bundle and click on "open". The first time won't let you open it and show you some kind of warning, but the second time it should be fine.

At the time of the last Mu release I think there was some dependencies bundled that were not High Sierra compatible, we've fixed the issue in the project, but haven't made a release yet.

mac-arno, 2024-01-11:
Thank you very much for your work. Now Mu 1.2.2 starts as it should. Menu Entries look a little bit unusual, but it runs without errors. Now you can safely note in the System Requirements section of the documentation, that v 1.2.2 runs on High Sierra.

carlosperate, 2024-01-11:
I'm glad it works @mac-arno, thanks for the initial report and testing the latest installer!
I'll close this as resolved and we'll update the documentation together with the next release 👍

closed 2024-01-11 by carlosperate