mu-editor/mu #2473: ERROR: No matching distribution found for pup

roboticsware, 2023-11-25:
### What were you trying to do?

make win64

### What steps did you take to trigger the issue?

1. Made virtual env with Python 3.8 by minoconda
2. Installed dependancies by the command, pip install -e ".[dev]"
3. Tried to build win64 version

### What did you expect to happen?

No build error

### What actually happened?

subprocess.CalledProcessError: Command '['./venv-pup/Scripts/pip.exe', 'install', 'pup']' returned non-zero exit status 1.

### Operating System Version

Windows 10

### Mu Version

1.2

### Editor Log

(mu) C:\dev\mu>python make.py win64
Fetching wheels
Downloading wheels to C:\dev\mu\mu\wheels\1.2.2.zip
Removing wheel/sdist files from C:\dev\mu\mu\wheels
Removing existing wheel/sdist C:\dev\mu\mu\wheels\1.2.2.zip
Running pip download for pgzero / ('pgzero>=1.2.1', 'pygame<2.1.3') / [] / []
Looking in links: c:\Users\JJ\AppData\Local\Temp\tmp4yt8al82
Collecting pgzero>=1.2.1
  Using cached pgzero-1.2.1-py3-none-any.whl (71 kB)
Collecting pygame<2.1.3
  Using cached pygame-2.1.2-cp38-cp38-win_amd64.whl (8.4 MB)
Collecting numpy (from pgzero>=1.2.1)
  Using cached numpy-1.24.4-cp38-cp38-win_amd64.whl.metadata (5.6 kB)
Using cached numpy-1.24.4-cp38-cp38-win_amd64.whl (14.9 MB)
Saved c:\users\jj\appdata\local\temp\tmp4yt8al82\pgzero-1.2.1-py3-none-any.whl
Saved c:\users\jj\appdata\local\temp\tmp4yt8al82\pygame-2.1.2-cp38-cp38-win_amd64.whl
Saved c:\users\jj\appdata\local\temp\tmp4yt8al82\numpy-1.24.4-cp38-cp38-win_amd64.whl
Successfully downloaded pgzero pygame numpy
Running pip download for flask / ('flask==2.0.3', 'Werkzeug<3.0.0') / [] / []
Looking in links: c:\Users\JJ\AppData\Local\Temp\tmp4yt8al82
Collecting flask==2.0.3
  Using cached Flask-2.0.3-py3-none-any.whl (95 kB)
Collecting Werkzeug<3.0.0
  Using cached werkzeug-2.3.8-py3-none-any.whl.metadata (4.1 kB)
Collecting Jinja2>=3.0 (from flask==2.0.3)
  Using cached Jinja2-3.1.2-py3-none-any.whl (133 kB)
Collecting itsdangerous>=2.0 (from flask==2.0.3)
  Using cached itsdangerous-2.1.2-py3-none-any.whl (15 kB)
Collecting click>=7.1.2 (from flask==2.0.3)
  Using cached click-8.1.7-py3-none-any.whl.metadata (3.0 kB)
Collecting MarkupSafe>=2.1.1 (from Werkzeug<3.0.0)
  Using cached MarkupSafe-2.1.3-cp38-cp38-win_amd64.whl.metadata (3.1 kB)
Collecting colorama (from click>=7.1.2->flask==2.0.3)
  Using cached colorama-0.4.6-py2.py3-none-any.whl (25 kB)
Using cached werkzeug-2.3.8-py3-none-any.whl (242 kB)
Using cached click-8.1.7-py3-none-any.whl (97 kB)
Using cached MarkupSafe-2.1.3-cp38-cp38-win_amd64.whl (17 kB)
Saved c:\users\jj\appdata\local\temp\tmp4yt8al82\flask-2.0.3-py3-none-any.whl
Saved c:\users\jj\appdata\local\temp\tmp4yt8al82\werkzeug-2.3.8-py3-none-any.whl
Saved c:\users\jj\appdata\local\temp\tmp4yt8al82\click-8.1.7-py3-none-any.whl
Saved c:\users\jj\appdata\local\temp\tmp4yt8al82\itsdangerous-2.1.2-py3-none-any.whl
Saved c:\users\jj\appdata\local\temp\tmp4yt8al82\jinja2-3.1.2-py3-none-any.whl
[... 164 lines cut ...]
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the 
SSL module is not available.")': /simple/pup/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the 
SSL module is not available.")': /simple/pup/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the 
SSL module is not available.")': /simple/pup/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the 
SSL module is not available.")': /simple/pup/
Could not fetch URL https://pypi.org/simple/pup/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/pup/ (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available.")) - skipping
ERROR: Could not find a version that satisfies the requirement pup (from versions: none)
ERROR: No matching distribution found for pup
WARNING: pip is configured with locations that require TLS/SSL, however the ssl module in Python is not available.
Could not fetch URL https://pypi.org/simple/pip/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/pip/ (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available.")) - skipping
Traceback (most recent call last):
  File "make.py", line 501, in <module>
    sys.exit(main(*sys.argv[1:]))
  File "make.py", line 497, in main
    return function(*args)
  File "make.py", line 453, in win64
    _build_windows_msi(bitness=64)
  File "make.py", line 439, in _build_windows_msi
    subprocess.check_call(cmd)
  File "C:\Users\JJ\miniconda3\envs\mu\lib\subprocess.py", line 364, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['./venv-pup/Scripts/pip.exe', 'install', 'pup']' returned non-zero exit status 1.

carlosperate, 2023-12-11:
Thanks for the report @roboticsware, just in case this happened to be an intermitent connection error, have you tried again since the issue was opened? Do you still get the same error?

roboticsware, 2024-01-01:
I fixed it by https://github.com/conda/conda/issues/8273.

closed 2024-01-01 by roboticsware