mu-editor/mu #2484: ImportError: Symbol not found: ____chkstk_darwin; Also unable to write venv.json

Kallaste, 2023-12-27:
I am getting repeated crashing on first run attempts of Mu Editor v.1.2.0 on MacOS 10.13.6. I strongly suspect it is a compatibility issue, but upgrading this machine to a newer MacOS is not an option, so I am hoping for a workaround. Older versions of Mu will run on this machine, but will not work with the device I am flashing.

The application simply crashes when clicked, but here is the error it throws when run from the terminal: 

```
No settings file found at /Users/tatiana/Library/Application Support/mu/venv.json; skipping 
Traceback (most recent call last):
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/__main__.py", line 1, in <module>
    from mu.app import run
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/app.py", line 48, in <module>
    from .modes import (
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/modes/__init__.py", line 1, in <module>
    from .python3 import PythonMode
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/modes/python3.py", line 27, in <module>
    from qtconsole.manager import QtKernelManager
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/qtconsole/manager.py", line 9, in <module>
    from jupyter_client import KernelManager
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/jupyter_client/__init__.py", line 4, in <module>
    from .connect import *
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/jupyter_client/connect.py", line 21, in <module>
    import zmq
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/zmq/__init__.py", line 103, in <module>
    from zmq import backend
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/zmq/backend/__init__.py", line 31, in <module>
    raise original_error from None
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/zmq/backend/__init__.py", line 26, in <module>
    _ns = select_backend(first)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/zmq/backend/select.py", line 31, in select_backend
    mod = import_module(name)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/zmq/backend/cython/__init__.py", line 6, in <module>
    from . import (
ImportError: dlopen(/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/zmq/backend/cython/_device.cpython-38-darwin.so, 2): Symbol not found: ____chkstk_darwin
  Referenced from: /Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/zmq/backend/cython/../../.dylibs/libsodium.23.dylib (which was built for Mac OS X 10.15)
  Expected in: /usr/lib/libSystem.B.dylib
 in /Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/zmq/backend/cython/../../.dylibs/libsodium.23.dylib
Unable to write settings to /Users/tatiana/Library/Application Support/mu/venv.json:
{
  "dirpath": "/Users/tatiana/Library/Application Support/mu/mu_venv-38-20231226-173033"
}
Traceback (most recent call last):
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/settings.py", line 181, in save
    with open(saving_to_filepath, "w", encoding="utf-8") as f:
FileNotFoundError: [Errno 2] No such file or directory: '/Users/tatiana/Library/Application Support/mu/venv.json'
```

carlosperate, 2023-12-27:
Thanks for the report @Kallaste, could you attach the full log here?
You should be able to find it in `~/Library/Logs/mu/mu.log`

Kallaste, 2023-12-27:
Thanks for the response. I have attached the log.

[mu.log](https://github.com/mu-editor/mu/files/13780897/mu.log)