mu-editor/mu #2489: The WIN32 version is made by a local build of my personal WIN 10 platform is not opened.

JeongJun-Lee, 2024-01-03:
### What actually happened?

I made this WIN32 personal build in conda virtual env with python 3.7.9 in 64bits WIN10 and installed it to my local PC and executed it for the first time in offline and met this crash error.

Do I need a 32bits WIN machine for WIN32 build?

### Operating System Version

Win 10

### Mu Version

1.2.2

### Editor Log

https://github.com/crashreport/issues/580

carlosperate, 2024-01-03:
Hi @JeongJun-Lee,

We are not really supporting 32-bit version of Window builds, as Microsoft stopped creating Windows 32-bit builds a while ago (talking specifically about the Windows builds, of course 32-bit applications are still officially supported in 64-bit Windows though) and since windows 7 the vast majority of users have been running 64-bit Windows.

Is your local machine a 32-bit build of Windows? Or was the 32-bit Mu installer failing in a 64 bit Windows build?

JeongJun-Lee, 2024-01-03:
Hi @carlosperate,

I built 32-bit Mu Windows version successfully in my 64-bit Windows 10 and installation of the built version was successful in my PC as well. But when I opened the Mu for the first time after the installation, the Mu spit out the crash error report in the middle of the executing.

As we can see the last line of the log: "ERROR: MarkupSafe-2.1.3-cp37-cp37m-win_amd64.whl is not a supported wheel on this platform.", I think the built version was packaged with wrong Python package bundle which is including 64-bit packages, not 32-bit packages only.

Eventually, I need to build 32-bit Mu Windows version for students in our country because most of them are still in Windows 7, unfortunately. Thanks.

carlosperate, 2024-01-04:
Right, yes, as we haven't been building 32 bit installers for a long time, the `win32` build target has not been maintained and not currently supported.

The main issue is that pup (the packager) will download wheels valid for the platform that is running, so if the installer is built in a 64 build of Windows it will download wheels for 64 bit Windows.
The simplest thing might be to build the installer in a 32-bit version of Windows, I cannot guarantee it'll work, but it should at least download the right wheel versions.

Alternatively you can have a look at how the mac build is passing a "platform" flag to specify the wheel platform, so you might be able to do something similar for `win32`.

JeongJun-Lee, 2024-03-25:
Thank for your advice.
I found a solution to make env for win32 build by Conda in 64 bits machine. And it works. So I'll close this.
You can see HowTo on the link: https://www.youtube.com/watch?v=o1Zzw-Y_n2g

closed 2024-03-25 by JeongJun-Lee