mu-editor/mu #2498: Python tkinter applications fail with "append_pending_request: Assertion `!xcb_xlib_unknown_seq_number' failed."

AlbertWeichselbraun, 2024-01-30:
### What were you trying to do?

Start a python application which creates a dialog box with turtle.textinput (i.e., tkinter.simpledialog)

### What steps did you take to trigger the issue?

Start the following code with Mu:

```python
from turtle import textinput
a = textinput("Title", "Input")
``` 

### What did you expect to happen?

A textinput field with title "Title" should ask the user for "Input".

### What actually happened?

The application crashes with the following output:

```
[xcb] Unknown sequence number while appending request
[xcb] You called XInitThreads, this is not your fault
[xcb] Aborting, sorry about that.
python: ../../src/xcb_io.c:157: append_pending_request: Assertion `!xcb_xlib_unknown_seq_number' failed.

---------- FINISHED ----------
exit code: 6 status: 1
```

### Operating System Version

Ubuntu 23.10 (Mantic)

### Mu Version

1.2.0

### Other Info

- Note that the same code works when the file is directly called with python or run with another IDE such as PyCharms.
- `echo $XDG_SESSION_TYPE` yields `wayland`

### Editor Log

2024-01-30 07:03:00,289 - root:283(run) INFO: 

-----------------

Starting Mu 1.2.0
2024-01-30 07:03:00,289 - root:284(run) INFO: uname_result(system='Linux', node='immanuel.fhgr.ch', release='6.5.0-14-generic', version='#14-Ubuntu SMP PREEMPT_DYNAMIC Tue Nov 14 14:59:49 UTC 2023', machine='x86_64', processor='x86_64')
2024-01-30 07:03:00,293 - root:285(run) INFO: Platform: Linux-6.5.0-14-generic-x86_64-with-glibc2.2.5
2024-01-30 07:03:00,293 - root:286(run) INFO: Python path: ['/tmp/.mount_Mu_Ediotdi4W/usr/lib/python38.zip', '/tmp/.mount_Mu_Ediotdi4W/usr/lib/python3.8', '/tmp/.mount_Mu_Ediotdi4W/usr/lib/python3.8/lib-dynload', '/tmp/.mount_Mu_Ediotdi4W/usr/lib/python3.8/site-packages']
2024-01-30 07:03:00,293 - root:287(run) INFO: Language code: en_US
2024-01-30 07:03:00,294 - mu.settings:220(load) WARNING: No settings file found at /home/albert/.local/share/mu/settings.json; skipping
2024-01-30 07:03:00,295 - root:312(run) INFO: Wayland detected, setting QT_QPA_PLATFORM=wayland
2024-01-30 07:03:00,817 - mu.virtual_environment:619(ensure_and_create) INFO: Added log handler.
2024-01-30 07:03:00,817 - mu.virtual_environment:630(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2024-01-30 07:03:00,818 - mu.virtual_environment:719(ensure_path) INFO: Virtual Environment found at: /home/albert/.local/share/mu/mu_venv-38-20240130-064515
2024-01-30 07:03:00,818 - mu.virtual_environment:740(ensure_interpreter) INFO: No interpreter found at: /home/albert/.local/share/mu/mu_venv-38-20240130-064515/bin/python
2024-01-30 07:03:00,819 - mu.virtual_environment:750(ensure_interpreter) INFO: Symlinked /home/albert/.local/share/mu/mu_venv-38-20240130-064515/bin/python to AppImage's /tmp/.mount_Mu_Ediotdi4W/usr/bin/python3.8
2024-01-30 07:03:00,819 - mu.virtual_environment:463(run_subprocess) INFO: Running ('/home/albert/.local/share/mu/mu_venv-38-20240130-064515/bin/python', '-c', 'import sys; print("%s%s" % sys.version_info[:2])') with kwargs {'shell': False}
2024-01-30 07:03:00,941 - mu.virtual_environment:475(run_subprocess) DEBUG: Process returned 0; output: 38
2024-01-30 07:03:00,941 - mu.virtual_environment:786(ensure_interpreter_version) INFO: Both interpreters at version 38
2024-01-30 07:03:00,942 - mu.virtual_environment:815(ensure_pip) INFO: Pip found at: /home/albert/.local/share/mu/mu_venv-38-20240130-064515/bin/pip
2024-01-30 07:03:00,942 - mu.virtual_environment:798(ensure_key_modules) DEBUG: Verifying import of: pgzero
2024-01-30 07:03:00,942 - mu.virtual_environment:463(run_subprocess) INFO: Running ('/home/albert/.local/share/mu/mu_venv-38-20240130-064515/bin/python', '-c', 'import pgzero') with kwargs {'shell': False}
2024-01-30 07:03:01,021 - mu.virtual_environment:475(run_subprocess) DEBUG: Process returned 0; output: No output received.
2024-01-30 07:03:01,021 - mu.virtual_environment:798(ensure_key_modules) DEBUG: Verifying import of: flask
2024-01-30 07:03:01,021 - mu.virtual_environment:463(run_subprocess) INFO: Running ('/home/albert/.local/share/mu/mu_venv-38-20240130-064515/bin/python', '-c', 'import flask') with kwargs {'shell': False}
2024-01-30 07:03:01,838 - mu.virtual_environment:475(run_subprocess) DEBUG: Process returned 0; output: No output received.
2024-01-30 07:03:01,839 - mu.virtual_environment:798(ensure_key_modules) DEBUG: Verifying import of: ipykernel
2024-01-30 07:03:01,839 - mu.virtual_environment:463(run_subprocess) INFO: Running ('/home/albert/.local/share/mu/mu_venv-38-20240130-064515/bin/python', '-c', 'import ipykernel') with kwargs {'shell': False}
2024-01-30 07:03:03,342 - mu.virtual_environment:475(run_subprocess) DEBUG: Process returned 0; output: No output received.
2024-01-30 07:03:03,342 - mu.virtual_environment:798(ensure_key_modules) DEBUG: Verifying import of: ipython_genutils
2024-01-30 07:03:03,342 - mu.virtual_environment:463(run_subprocess) INFO: Running ('/home/albert/.local/share/mu/mu_venv-38-20240130-064515/bin/python', '-c', 'import ipython_genutils') with kwargs {'shell': False}
2024-01-30 07:03:03,427 - mu.virtual_environment:475(run_subprocess) DEBUG: Process returned 0; output: No output received.
2024-01-30 07:03:03,428 - mu.virtual_environment:662(ensure_and_create) INFO: Valid virtual environment found at /home/albert/.local/share/mu/mu_venv-38-20240130-064515
2024-01-30 07:03:03,432 - mu.settings:169(save) DEBUG: Saving to /home/albert/.local/share/mu/venv.json
2024-01-30 07:03:03,480 - mu.logic:777(__init__) INFO: Setting up editor.
2024-01-30 07:03:03,488 - mu.logic:801(__init__) INFO: Log directory: /home/albert/.cache/mu/log
2024-01-30 07:03:03,489 - mu.logic:802(__init__) INFO: Data directory: /home/albert/.local/share/mu
2024-01-30 07:03:03,489 - mu.logic:816(setup) INFO: Available modes: python, snek, circuitpython, microbit, esp, web, pyboard, debugger, pygamezero, lego, pico
2024-01-30 07:03:03,490 - mu.modes.base:61(get_default_workspace) INFO: Using workspace /home/albert/mu_code from settings file
2024-01-30 07:03:03,654 - mu.modes.base:61(get_default_workspace) INFO: Using workspace /home/albert/mu_code from settings file
2024-01-30 07:03:03,655 - mu.logic:1583(change_mode) INFO: Workspace directory: /home/albert/mu_code
2024-01-30 07:03:03,655 - mu.logic:890(restore_session) DEBUG: <SessionSettings from /home/albert/.local/share/mu/session.json>
2024-01-30 07:03:03,674 - mu.logic:1015(_load) INFO: Loading script from: /home/albert/tmp/test.py
2024-01-30 07:03:03,675 - mu.logic:332(read_and_decode) DEBUG: Trying to decode with utf-8
2024-01-30 07:03:03,676 - mu.logic:335(read_and_decode) INFO: Decoded with utf-8
2024-01-30 07:03:03,677 - mu.logic:348(read_and_decode) DEBUG: Detected newline '\n'
2024-01-30 07:03:03,677 - mu.logic:1114(_load) DEBUG: # Write your code here :-)
from turtle import textinput
a = textinput("Example", "Application")

2024-01-30 07:03:04,218 - mu.logic:912(restore_session) INFO: Loaded files.
2024-01-30 07:03:04,218 - mu.logic:918(restore_session) INFO: User defined environment variables: {}
2024-01-30 07:03:04,218 - mu.logic:923(restore_session) INFO: Minify scripts on micro:bit? False
2024-01-30 07:03:04,285 - mu.modes.base:61(get_default_workspace) INFO: Using workspace /home/albert/mu_code from settings file
2024-01-30 07:03:04,285 - mu.logic:1583(change_mode) INFO: Workspace directory: /home/albert/mu_code
2024-01-30 07:03:06,785 - mu.modes.python3:230(run_script) INFO: About to run script: {'interpreter': '/home/albert/.local/share/mu/mu_venv-38-20240130-064515/bin/python', 'script_name': '/home/albert/tmp/test.py', 'working_directory': '/home/albert/tmp', 'interactive': True, 'envars': {}}
2024-01-30 07:03:06,810 - mu.interface.main:780(add_python3_runner) INFO: About to start_process: '/home/albert/.local/share/mu/mu_venv-38-20240130-064515/bin/python', '/home/albert/tmp/test.py', '/home/albert/tmp', True, False, None, {}, None
2024-01-30 07:03:06,810 - mu.interface.panes:1073(start_process) INFO: Running script: /home/albert/tmp/test.py
2024-01-30 07:03:06,811 - mu.interface.panes:1074(start_process) INFO: Using interpreter: /home/albert/.local/share/mu/mu_venv-38-20240130-064515/bin/python
2024-01-30 07:03:06,811 - mu.interface.panes:1076(start_process) INFO: Running with interactive mode.
2024-01-30 07:03:06,811 - mu.interface.panes:1079(start_process) INFO: Command args: []
2024-01-30 07:03:06,811 - mu.interface.panes:1101(start_process) INFO: Working directory: /home/albert/tmp
2024-01-30 07:03:06,812 - mu.interface.panes:1105(start_process) INFO: Python path: ['/tmp/.mount_Mu_Ediotdi4W/usr/lib/python38.zip', '/tmp/.mount_Mu_Ediotdi4W/usr/lib/python3.8', '/tmp/.mount_Mu_Ediotdi4W/usr/lib/python3.8/lib-dynload', '/tmp/.mount_Mu_Ediotdi4W/usr/lib/python3.8/site-packages']
2024-01-30 07:03:06,812 - mu.interface.panes:1135(start_process) INFO: Args: ['-i', '/home/albert/tmp/test.py']

carlosperate, 2024-01-30:
Thanks for the report @AlbertWeichselbraun, is this using the AppImage version of Mu?

AlbertWeichselbraun, 2024-01-31:
Yes,  I am using the official AppImage.