mu-editor/mu #2507: MU incredibly slow with print

rebtop, 2024-03-17:
It displays 3 - 4 characters every 2 - 3 minutes

Cntl C takes 5 - 6 minutes to execute or more at times.

Even after i reject the MatrixPortal M4 from my Win 11 system, it keeps running and running and running....

Speedy-VI, 2024-04-14:
I am having the  same issue with the MU Serial Monitor when running code that is flashing 64 NeoPixels either red, blue, or off. The state of each LED is determined by a random number function and assignable probabilities. I added a print(Selected Color:", Color) line so I could see what colors the selection logic is outputting. The 64 LEDs are changing color pretty fast and the Serial Monitor is not keeping up. The Serial Monitor freezes and so do the LEDs. MU gets slow and jerky and reports a COM problem. Sometimes MU freezes and I just see a spinny wheel. I tried enabling the Serial Plotter and immediately got a Data Flood Detected popup.  I am sure this is due to how fast the LEDS are changing color but I don't recall the Arduino IDE Serial Monitor having this problem. It will happily scroll lines far faster than any human could read them, and it does not cause the program to freeze. I don't see any way to change the baud rate. I can live with the Serial Monitor not keeping up with every LED color change but when it happens it also seems to stop the code because the LEDS also freeze.

rebtop, 2024-04-14:
I stopped using MU. Unstable Sent from my iPhone