mu-editor/mu #2510: Does not detect my working circuitpython board which uses serial only

gitcnd, 2024-05-09:
Thanks for taking the time to fill out this bug report!

Please answer each question below to the best of your ability.
It's okay to leave some blank if it doesn't apply to your problem.

What were you trying to do?
Upload and run circuitpython code
What steps did you take to trigger the issue?
Install and run
What did you expect to happen?
Ask me for the serial port to use, then use it
What actually happened?
told me it did not detect the drive, which is correct since there is no drive, but wrong, because it assumed there should be, which is not the case for ESP32 classic boards which use serial chips.  They behave like micropython ones and use serial instead of USB drive emulation for file operations.
Operating System Version
Windows
Mu Version
1.2
Other Info
You can put any info here that you think might help us track down the bug.

Comment
There was an error creating your Issue: body is too long (maximum is 65536 characters).
The serial REPL did work in circuitpython, weirdly, but there's no way to list the files.  The serial REPL does NOT work in micropython mode.

says this on startup:-

[image]

(micropython, although looking more promising, also does not work, because you're wrongly setting the hardware flow control, which holds my ESP32 board in reset mode)
 
Editor Log
Please remember to attach a copy of the full log files for Mu. You
can get the logs by clicking on the cog icon in the bottom right of the
editor window.

2024-05-09 10:18:44,003 - root:283(run) INFO: 

-----------------

Starting Mu 1.2.0
2024-05-09 10:18:44,003 - root:284(run) INFO: uname_result(system='Windows', node='RoG', release='10', version='10.0.17763', machine='AMD64', processor='Intel64 Family 6 Model 151 Stepping 2, GenuineIntel')
2024-05-09 10:18:44,043 - root:285(run) INFO: Platform: Windows-10-10.0.17763-SP0
2024-05-09 10:18:44,043 - root:286(run) INFO: Python path: ['C:\\Users\\cnd\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\cnd\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\cnd\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\cnd\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\cnd\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\cnd\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\cnd\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\cnd\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin']
2024-05-09 10:18:44,043 - root:287(run) INFO: Language code: en_AU
2024-05-09 10:18:44,043 - mu.settings:220(load) WARNING: No settings file found at C:\Users\cnd\AppData\Local\python\mu\settings.json; skipping
2024-05-09 10:18:44,043 - mu.settings:220(load) WARNING: No settings file found at C:\Users\cnd\AppData\Local\python\mu\session.json; skipping
2024-05-09 10:18:44,186 - mu.virtual_environment:619(ensure_and_create) INFO: Added log handler.
2024-05-09 10:18:44,442 - mu.virtual_environment:630(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2024-05-09 10:18:44,443 - mu.virtual_environment:837(create_venv) INFO: Creating virtualenv: 

[etc] "Comment is too long (maximum is 65536 characters)"

2024-05-09 10:29:57,281 - mu.virtual_environment:189(wait) DEBUG: Finished: True; exitStatus 0; exitCode 0
2024-05-09 10:29:57,281 - mu.virtual_environment:274(run) DEBUG: Process output: Package           Version
----------------- -------
asttokens         2.1.0
backcall          0.2.0
click             8.1.3
colorama          0.4.6
decorator         5.1.1
entrypoints       0.4
executing         1.2.0
Flask             2.0.3
ipykernel         5.5.6
ipython           8.6.0
ipython-genutils  0.2.0
itsdangerous      2.1.2
jedi              0.18.1
Jinja2            3.1.2
jupyter_client    7.4.5
jupyter_core      5.0.0
MarkupSafe        2.1.1
matplotlib-inline 0.1.6
nest-asyncio      1.5.6
numpy             1.23.4
parso             0.8.3
pgzero            1.2.1
pickleshare       0.7.5
pip               22.3.1
platformdirs      2.5.4
prompt-toolkit    3.0.32
pure-eval         0.2.2
pygame            2.1.2
Pygments          2.13.0
python-dateutil   2.8.2
pywin32           305
pyzmq             24.0.1
setuptools        65.5.1
six               1.16.0
stack-data        0.6.1
tornado           6.2
traitlets         5.5.0
wcwidth           0.2.5
Werkzeug          2.2.2
wheel             0.38.4
2024-05-09 10:29:57,281 - mu.virtual_environment:993(installed_packages) INFO: []

carlosperate, 2024-06-04:
Thanks for the report @gitcnd. Do you have a bit more info about the board you are using? Specially the ESP model number?

It's possible this model is newer that what was supported by the esptool version currently included with Mu.

gitcnd, 2024-06-05:
I expect it's at least all of these: https://circuitpython.org/downloads?mcufamilies=esp32 (since the "classic" esp32 doesn't support the USBC magic of the new boards).  In my case, it was https://circuitpython.org/board/ai-thinker-esp32-cam/ , https://circuitpython.org/board/esp32-wrover-dev-cam/ and https://templates.blakadder.com/wemos_D1_Mini_ESP32.html

The bug is a logic error in your code - expecting a drive to show up, when none ever will (even though your code can work with serial-only, and even though the serial REPL gives access to all files and folders anyhow...)

carlosperate, 2024-12-16:
Oh right, so was the issue that in the ESP MicroPython mode or the CircuitPython mode?
Unfortunately in CircuitPython mode we currently only support CircuitPython boards that mount a USB MSD drive, as the main way to load the code is by writing a code.py file into it.

However, I would expect the ESP MicroPython mode to still work (even if you have to constantly cancel Mu trying to switch modes). Did that not work?