mu-editor/mu #2511: cannot report bugs

gitcnd, 2024-05-09:
form requires us to paste in the logs, but doing that causes the form not to allow the bug to be submitted because the logs make the thing say this:

      Comment is too long (maximum is 65536 characters)

carlosperate, 2024-06-03:
Hi @gitcnd, the best way to attach the log to an issue is to drag and drop the file into the textbox, that way it is attached as a .txt file and doesn't have the same kind of limitations.

gitcnd, 2024-06-04:
or you could fix the form

carlosperate, 2024-06-04:
Unfortunately the form limit is a GitHub feature controlled by GitHub, so there isn't anything we can do about that.
As the question has been answered I'll close this, but please don't hesitate to let us know if you are encountering any issues with the Mu editor.

closed 2024-06-04 by carlosperate