mu-editor/mu #2538: Microbit identified by not flashed on MacOS Sequoia 15.1.1 (24B91)

deejaygraham, 2024-12-13:
### What were you trying to do?

Flash a program to an attached microbit

### What steps did you take to trigger the issue?

After upgrading MacOS from Big Sur and using the latest mu editor, attached microbit USB port. Port shows that microbit is attached and bottom left of editor shows that it identifies that a microbit has been detected ("Detected new microbit device"). Pressed the flash button.

### What did you expect to happen?

Program to be flashed to device 

### What actually happened?

 Pressing "flash" brings up the file dialog and cancelling shows the usual "Could not find an attached microbit" message. 

### Operating System Version

MacOS Sequoia 15.1.1

### Mu Version

1.2.0

### Other Info

Made sure that mu and Terminal has full access to disk under Settings > Privacy and Security. Tried the same procedure with another machine running Big Sur and this flashed as expected.

### Editor Log

2024-12-13 09:43:15,577 - mu.logic:756(check_usb) INFO: microbit device connected on port: /dev/cu.usbmodem102(VID: 0x0D28, PID: 0x0204, manufacturer: 'ARM')
2024-12-13 09:43:46,303 - mu.modes.microbit:273(flash) INFO: Preparing to flash script.
2024-12-13 09:43:46,305 - mu.modes.microbit:280(flash) DEBUG: Python script from 'snowflake.py' tab:
2024-12-13 09:43:46,305 - mu.modes.microbit:281(flash) DEBUG: b'# Snowglobe\nfrom microbit import *\nimport random\n\n\nclass Snowflake:\n    def __init__(self, x, y, brightness):\n        self.x = x\n        self.y = y\n        self.brightness = brightness\n\n    def draw(self):\n        display.set_pixel(self.x, self.y, 0)\n        display.y = min(display.y + 1, 4)\n        display.set_pixel(self.x, self.y, self.brightness)\n\n\nwidth = 4\nheight = 4\n\nsnowflake = Image("00900:00000:00000:00000:00900")\nsettledSnow = Image("00000:00000:00000:00000:99999")\ndisplay.show(snowflake)\nsleep(2000)\n\nsnowflakes = []\n\nfor i in range(1):\n    snowflakes.append(Snowflake(random.randint(0, 4), -1, random.randint(1, 9)))\n\nwhile True:\n\n    for i in range(1):\n        snowflakes[i].display()\n\n    # if accelerometer.was_gesture("shake"):\n    # help   display.clear()\n    # generate showflakes\n    # track each one as it falls\n'
2024-12-13 09:43:46,939 - mu.modes.microbit:229(find_microbit) INFO: Path to micro:bit: None
2024-12-13 09:43:46,939 - mu.modes.microbit:235(find_microbit) INFO: Serial port: /dev/cu.usbmodem102
2024-12-13 09:43:46,939 - mu.modes.microbit:236(find_microbit) INFO: Device serial number: 9904360254174e45004600100000003c000000009796990b
2024-12-13 09:43:46,939 - mu.modes.microbit:237(find_microbit) INFO: Board ID: 0x9904
2024-12-13 09:43:52,210 - mu.interface.main:457(get_microbit_path) DEBUG: Getting micro:bit path: 
2024-12-13 09:43:52,210 - mu.modes.microbit:299(flash) DEBUG: User defined path to micro:bit: 
2024-12-13 09:43:52,245 - mu.interface.main:1047(show_message) DEBUG: Could not find an attached BBC micro:bit.
2024-12-13 09:43:52,245 - mu.interface.main:1048(show_message) DEBUG: Please ensure you leave enough time for the BBC micro:bit to be attached and configured correctly by your computer. This may take several seconds. Alternatively, try removing and re-attaching the device or saving your work and restarting Mu if the device remains unfound.
2024-12-13 09:46:23,147 - mu.logic:1439(show_admin) INFO: Showing admin with logs from /Users/derekgraham/Library/Logs/mu/mu.log
2024-12-13 09:46:23,148 - mu.virtual_environment:984(installed_packages) INFO: Discovering installed third party modules in venv.
2024-12-13 09:46:23,151 - mu.virtual_environment:139(run_blocking) INFO: About to run blocking /Users/derekgraham/Library/Application Support/mu/mu_venv-38-20241210-084055/bin/pip with args ['list', '--disable-pip-version-check'] and envvars {}
2024-12-13 09:46:23,755 - mu.virtual_environment:189(wait) DEBUG: Finished: True; exitStatus 0; exitCode 0
2024-12-13 09:46:23,756 - mu.virtual_environment:274(run) DEBUG: Process output: Package           Version
----------------- -------
appnope           0.1.3
asttokens         2.1.0
backcall          0.2.0
click             8.1.3
decorator         5.1.1
entrypoints       0.4
executing         1.2.0
Flask             2.0.3
ipykernel         5.5.6
ipython           8.6.0
ipython-genutils  0.2.0
itsdangerous      2.1.2
jedi              0.18.1
Jinja2            3.1.2
jupyter_client    7.4.5
jupyter_core      5.0.0
MarkupSafe        2.1.1
matplotlib-inline 0.1.6
nest-asyncio      1.5.6
numpy             1.23.4
parso             0.8.3
pexpect           4.8.0
pgzero            1.2.1
pickleshare       0.7.5
pip               22.1.2
platformdirs      2.5.4
prompt-toolkit    3.0.32
ptyprocess        0.7.0
pure-eval         0.2.2
pygame            2.1.2
Pygments          2.13.0
python-dateutil   2.8.2
pyzmq             24.0.1
setuptools        62.6.0
six               1.16.0
stack-data        0.6.1
tornado           6.2
traitlets         5.5.0
wcwidth           0.2.5
Werkzeug          2.2.2
wheel             0.37.1
2024-12-13 09:46:23,756 - mu.virtual_environment:993(installed_packages) INFO: []

carlosperate, 2024-12-16:
Unfortunately an update in macOS has caused the MICROBIT drive to be missnamed as `NO NAME`, so Mu probably won't be able to find it because of that:
- https://github.com/ARMmbed/DAPLink/issues/1082

Apple has been informed, but we don't know when it might be fixed. In the meantime a micro:bit firmware update will be released soon, but that is currently being developed.

deejaygraham, 2024-12-17:
Thank you for the update

maxxlife, 2025-01-24:
Confirm issue still present. Mac OS 15.2 (24C101)