mu-editor/mu #2541: Failed to write into a file I created

Orto71, 2025-01-09:
### What were you trying to do?

Write into a file I created

### What steps did you take to trigger the issue?

This is the code I tried to run

sales = open('file_name.txt', 'w')
product=""
while not product=="Basta":
    product = input('Product name: ')
    if not product=="Basta":
        quantity = input('Product sales quantity for the day: ')
        sales.write(product + ',' + quantity + '\n')
sales.close

These are the data I tried to write into the file

coffee,25
cappuccino,34
flat white,12

### What did you expect to happen?

I expected to find some data in "file_name.txt"

### What actually happened?

Nothing.
The file is empty.
Now, because I'm a newbie in programming, I start by thinking it's my fault.
However, when I run the program in debug mode, it writes down into the file exactly what I want.
Therefore, please don't get mad at me if it's my fault, because this is really weird.

### Operating System Version

Windows 11

### Mu Version

1.2

### Editor Log

[mu.log](https://github.com/user-attachments/files/18363231/mu.log)

dhalbert, 2025-01-09:
This is a Python question, not a Mu problem. You should ask in a Python support forum of some kind.

Having said that, you should be doing `sales.close()` instead of `sales.close`.

Orto71, 2025-01-09:
Thanks.
Still it doesn't make sense that, with no brackets, the open method works in debug mode and it doesn't in run mode.
Moreover, why not putting any brackets didn't raise any error whatsoever?
Anyway, I really appreciate your answer.