mu-editor/mu #2566: When code.py crashes - debug window contents disappear

Hagtronics, 2025-07-25:
### What were you trying to do?

Edit code

### What steps did you take to trigger the issue?

Found that these line of code will work,
import board
import rotaryio
tuning = rotaryio.IncrementalEncoder(board.GP1, board.GP2, divisor=4)

But these line cause everything to crash,
import board
import rotaryio
tuning = rotaryio.IncrementalEncoder(board.GP1, board.GP11, divisor=4)

(root cause is that the encoder pins have to be sequential which is not documented anywhere - I have opened an issue with CircuitPython)

As for Mu behavior: First there are some debug messages which are quickly gone as the serial window closes - it seems this error hard resets the PICO and USB serial is lost. Then the device / editor is in safe mode that can only be exited by fixing the code.py file on the Pico, exiting the editor, unplugging the pico, then replugging the Pico, then restarting the editor.

It took a long time to finally see the error message 'flash by' before disappearing for me to see the issue. Making debugging near impossible.

If possible, it would be really helpful to log the last load messages from the serial window somewhere (I could not find them, if they are kept).

-OR- Other syntax errors seem to just stop the running program and I can fix them, for some reason this particular error crashes the whole shebang. Perhaps the error level/severity can be lowered so the editor does not crash. 

Thanks for your time and effort on the editor.

dhalbert, 2025-07-25:
This is a CircuitPython bug: https://github.com/adafruit/circuitpython/issues/10487

Hagtronics, 2025-07-27:
Hi dhalbert - yes root cause is Circuit Python - however my bringing it up was to suggest that Mu possibly could be more user friendly in the 'safe mode' handling. Specifically can Mu save / log the last contents of the serial window somewhere before it erases it? This bug was very hard to figure out when the traceback information was written and immediately disappears before it can be read. 

Thanks for your time and understanding.